pdx-robotics/Arduino_KHR-1 #4: Left/Right servo position translation.

Jordanb716, 2018-06-02:
Add a function that translates positions on one side of the body to mirror the same positions on the other side of the body.

Need to decide a standard on which directions should be 0 vs 180, could also use the opportunity to just swap that for certain joints if desired to make programming positions more intuitive.

@halyngoc Poke.

Jordanb716, 2018-06-02:
@halyngoc  No idea how to mark this feature request/bug, like you mentioned.

Chana030102, 2018-06-03:
It was brought to my attention that the **servo of the right shoulder pitch goes too far inward** to the side KHR-1. Test to see if it can move in the same matter as the left shoulder pitch. There is potentially a centering issue, which means the servo was not installed onto the chassis in the correct manner.

halyngoc, 2018-06-03:
@Chana030102 Go ahead and start a new issue on this since it's unrelated.

halyngoc, 2018-06-04:
I'll add an ArmPose constructor that converts all the angle arguments on the right to 180 - itself. Or should I do that to the left side?

Jordanb716, 2018-06-04:
I was thinking we should decide on some standard, like higher numbers are more forward/up/out in relation to the base pose described in the servo map document I made, and set each servo pair to match that standard.

halyngoc, 2018-06-04:
So since in https://github.com/pdx-robotics/Arduino_KHR-1/commit/9e7b13ea22868313652ab3702c6fcd299bc31d45 you only show angles for the right side, I'm assuming you want the left side to match that?

Jordanb716, 2018-06-04:
That's only supposed to map what the servos do unmodified, ideally if an angle was sent to both sides it should be mirrored if that's what you're asking.

I only show the right side in that document because I put at the top that all the servos on the left were the opposite, so I figured it would just add clutter to explicitly list them out.

halyngoc, 2018-06-04:
I understand what you're going for, I just need to know which side is the standard so I can convert the other side.

And I just realized that our waving functions will also need to change after this. I'm thinking of turning them into a series of ArmPoses. Maybe an extra class called Animation or something.

Jordanb716, 2018-06-04:
That's why I was saying we should come up with an overall standard, because whatever we standardize on might some of the directions from the left and some from the right. And I don't really want you to put in values until we decide on a standard because I'd hate for it to have to be redone in the future.

We could use forward/up/out as higher numbers, which would use the default for the right shoulder roll, but the left shoulder pitch. We need to decide on a standard first. We also need to do some more testing on the rest of the servo's to see what their defaults are, maybe we could ambush @Chana030102 in the lab sometime this week, or wait for summer?

halyngoc, 2018-06-21:
You should also update [Servo Direction Map.txt](https://github.com/pdx-robotics/Arduino_KHR-1/blob/feature/semaphore/Servo%20Direction%20Map.txt) to our new standard btw @Jordanb716

Jordanb716, 2018-07-10:
Servo direction map is now finished. (How do you link to files like that on here?) @halyngoc

halyngoc, 2018-08-14:
@Jordanb716 Will we also do the same thing for the legs?

halyngoc, 2018-08-26:
All conversions have been added and are working as expected, so I'll close this for now. Reopen if you notice anything weird.

closed 2018-08-26 by halyngoc