pdx-robotics/Arduino_KHR-1 #11: Create diagram of HR-SC04 with dimensions

Chana030102, 2018-07-21:
Make a diagram for the Ultrasonic Sensor with dimensions for the board. If a component on the board sticks out a significant amount, it should also be measured and documented on the diagram as well. Push diagram to 'hardware' branch.

jakechung42, 2018-10-09:
@Jordanb716 did you ever finish this? I remember seeing a blueprint for this sometimes ago.

Jordanb716, 2018-10-09:
I took photos with measurements, but then forgot to upload them iirc, and they died when my previous phone did.

jakechung42, 2018-10-09:
alright I'll make a model for the sensor. Should be easy to do.

jakechung42, 2018-10-10:
If anyone wants to learn 3D modeling, or measuring dimensions, I can use this assignment to help you. The HR-SC04 is simple enough that we can get it done easily and quickly.

Chana030102, 2018-10-10:
@jakechung42 this is low priority, so let's go ahead and leave this task as an opportunity for anyone new to 3D modeling.

williamVbrines, 2018-11-07:
I am going to close this because there is no current plan to use the HR-SC04.

closed 2018-11-07 by williamVbrines