pdx-robotics/Arduino_KHR-1 #12: Modify files for consistency in documentation

Chana030102, 2018-07-21:
Several things that we not planned out well in the beginning need to be changed for better consistency:
(1) Servo channel diagram reference image should match what's in code. 
(2) Change header and cpp files from "KHR_1" to "Arduino_KHR1" 
(3) Modify README.md accordingly

Chana030102, 2018-10-08:
@williamVbrines Please refer to my updated comment at the top and my comment on #31 about working on this issue. Thanks!

williamVbrines, 2018-10-08:
Thank you for the for the update I will make sure to make a branch off features/ semaphore and collaborate with @halyngoc in the future. However, who do I contact to review changes when making a pull request you or @halyngoc.

halyngoc, 2018-10-08:
Quoting [my comment](https://github.com/pdx-robotics/Arduino_KHR-1/pull/25#issuecomment-427985433) from #25:
> I've noticed that master is too outdated compared to this branch which makes working on other issues more difficult. So I'll look into getting this merged as soon as possible.

@williamVbrines, you can ask me or @Chana030102 to review your PR, but I'm not authorized to merge it.

halyngoc, 2018-10-09:
> - [ ] Servo channel diagram reference image should match what's in code. (cause that's what it's for)

> - [ ] Modify README.md accordingly

@Chana030102 I saw you already made commits related to these tasks. Can we check them off?

halyngoc, 2018-10-09:
I noticed just now but it's "Arduino_KHR-1" not "Arduino_KHR1"
I feel bad for @williamVbrines 😃

williamVbrines, 2018-10-09:
Don't worry got the header and cpp file changed to the right name now.

PR #32 (2018-10-09): Modify files for consistency 
This is a change to some files in the features/semaphore branch to help with consistency in the code.

Resolves #12.

Changes include:
- Renaming the following files
lib/KHR_1.h -> lib/Arduino_KHR-1.h
lib/KHR_1.cpp -> lib/Arduino_KHR-1.cpp
- Changing all #includes and references from lib/KHR_1.h to lib/Arduino_KHR1.h

As for changing the KHR-1_ServoChannels to match what is in the code, it has already been done and the README.md has been updated.
commit 18b5b9b0 (2018-10-09): Merge pull request #32 from pdx-robotics/Modify-files-for-constancy-

Changes have been reviewed and should not be problematic. Resolves #12
closed 2018-10-09 by Chana030102