pdx-robotics/Arduino_KHR-1 #18: Determine powering method and how to equip on KHR-1

Chana030102, 2018-07-31:
Determine if 4 AA batteries will be sufficient enough in powering all 17 servos.  Once a sufficient powering source is determined, figure out how KHR-1 will carry the power source.

Chana030102, 2018-08-16:
We tried a battery pack that rated at 6V and 750mah. It performed rather well and seemed to provide enough current to hold torque on 16 servos.
It consisted of five 1.2V batteries that can be found at the following website: [https://www.batteriesplus.com/battery/rechargeable/2_3a/nickel-cadmium]

Power method with AA batteries still needs to be experimented with as an alternative method.

jakechung42, 2018-10-09:
This is the one that we don't know how to charge it right?

Chana030102, 2018-10-09:
@jakechung42 This is to determine a powering method that doesn't require a power supply and a way to equip said method onto him so he can walk around with it.

jakechung42, 2018-10-09:
> 
> 
> @jakechung42 This is to determine a powering method that doesn't require a power supply and a way to equip said method onto him so he can walk around with it.

yeah but I'm asking about the battery. I remember we tested some battery but we couldn't figure out how to charge it.

Chana030102, 2018-10-09:
Yes, that's [noted in this comment](https://github.com/pdx-robotics/Arduino_KHR-1/issues/18#issuecomment-413575004). We haven't figured out the proper way to charge it (if there is a proper way at all).

jakechung42, 2018-10-09:
> 
> 
> > @jakechung42 This is to determine a powering method that doesn't require a power supply and a way to equip said method onto him so he can walk around with it.
> 
> yeah but I'm asking about the battery. I remember we tested some battery but we couldn't figure out how to charge it.

if that is the case that it is the batteries that I remember, I think I might look into how to charge it or to build a charger for it.

williamVbrines, 2018-10-10:
I suggest looking into this power bank if you can get away with a lower voltage and slightly higher voltage.
5V self-regulating, 2.4amp, and can be chared through USB.

https://www.bkstr.com/ProductDisplay?urlRequestType=Base&catalogId=10001&categoryId=36696&productId=76703340&errorViewName=ProductDisplayErrorView&langId=-1&top_category=10002&parent_category_rn=10002&storeId=381905

williamVbrines, 2018-10-27:
The stance on this issue has been changed so that having a tetherless way of powering KHR-1 is not a priority nor needed for KHR1. Thus a way of powering KHR1 has been devised by using an RTX Powersuply. 
This Issue will now be closed until the need arises for KHR1 to walk freely.

closed 2018-10-27 by williamVbrines