pdx-robotics/Arduino_KHR-1 #26: Busted Hip Servo Replacement

Chana030102, 2018-09-28:
One of the hip servos were damaged and did not maintain torque during the demo at Party in the Park. This servo (and any other broken servos) need to be replaced.

halyngoc, 2018-10-06:
Now there are 4 broken servos 👍

jakechung42, 2018-10-06:
since we know the issue is the servos now, gonna remove bug tag. Now just wait for the servos to be replace. I have to also test or the servos to make sure that not only those 4 are damages.

williamVbrines, 2018-10-09:
The servos that are broke are on. Check off when replaced.
[ ]CH3
[ ]CH6
[ ]CH9
[ ]Ch10

williamVbrines, 2018-10-11:
I tested all servos that are currently on KHR-1 with a custom program of mine all of them hold work.

jakechung42, 2018-10-18:
@williamVbrines did you reattached all the servos and rebuild KHR-1?

williamVbrines, 2018-10-18:
No is still don't have a working origin program that I can compile and load onto KHR1. I am waiting on ha for this to happen because I know she has a working program for doing this.

jakechung42, 2018-10-18:
Okay, we'll work on it tomorrow.

jakechung42, 2018-10-18:
opss clicked wrong button.

williamVbrines, 2018-10-20:
Someone close this issue.

closed 2018-10-21 by halyngoc