pdx-robotics/Arduino_KHR-1 #36: Make coding and documentation guides.

williamVbrines, 2018-10-11:
There have been issues that have popped up in the past like naming conventions, code portability, archiving, file management, and some documentation problems that need to be addressed, then changed. The files for these guidelines should be in the main directory of the repo as a text file.

These guidelines should include examples, does and don'ts of what to do. 

- [ ] Guidelines for Git. 
- [ ] Guidelines for the code.

williamVbrines, 2018-10-11:
We should discuss this at the meeting on Friday at some point.

halyngoc, 2018-10-11:
Agreed. We really need to establish style guides. And fix that god awful one we currently have 😃

halyngoc, 2018-10-11:
I'm working on a [practice repo](https://github.com/pdx-robotics/github-practice) where I basically reexplain [GitHub flow](https://guides.github.com/introduction/flow/) with some extra instructions on each step. The [wiki pages](https://github.com/pdx-robotics/github-practice/wiki) there might be a good place to put all these style guides. when we have them